ihormelnyk/opentherm_library #1: OpenTherm v3.0 Compatibility?

regnerus, 2018-06-12:
I understand this might be the wrong repository to ask such a question, however I was wondering if you have any information about compatibility with OpenTherm protocol specification v3.0. Unfortunately I can not find the specification information online but as I understand the signal low and signal high voltages are higher then with specification V2.2.

If I measure the voltage between the OpenTherm connections on my central heating unit/boiler I measure around 26.5V. Would it be possible to switch out the Zeners to higher value ones? Furthermore does anyone happen to know what the signal low and signal high values for V3.0 are? I assume the signal low will be around that 26.5V, but is there a method to also measure the signal high voltage?

Help would be greatly appreciated however please do not hesitate to close this issue since it is not directly related to the OpenTherm library.

ihormelnyk, 2018-06-12:
Hi @regnerus ,
I can't find OpenTherm v3.0 specification also, but I believe most likely these specifications are compatible by voltage levels. OpenTherm is low power protocol, so you can see higher values on unconnected wires. I think you will see valid voltage levels if you connect boiler to some opentherm thermostat or [adapter](http://ihormelnyk.com/opentherm_adapter)

regnerus, 2018-06-13:
Hi @ihormelnyk, thanks for the fast reply! I know that the specification is not online... I actually emailed the OpenTherm association about this however you have to become a (paid) member in order to see those documents.

I have the Honeywell Round Modulation thermostat which supports OpenTherm. I tried however to build an OpenTherm adapter to the exact schematic and components as provided on your page and validated that all the connections are working as supposed to, however I get a constant timeout message from the Arduino library. 

The wikipedia page about OpenTherm https://en.wikipedia.org/wiki/OpenTherm mentions OpenTherm V3.0 mentions something called OpenTherm Smart Power, to allow thermostats running of the power from the connection, which got me wondering if the actual voltages could differ in that case or if just the amperage is different. 

Currently I am just trying to explore all options to why the adapter I've built is not working...

ihormelnyk, 2018-06-13:
Hi @regnerus ,
What kind of controller do you use, what pins do you use for communication between controller and adapter? Are you sure that adapter output pin connected to controller input pin and adapter input pin to controller output pin and they are properly configured in sketch? Can you attach some photos?

regnerus, 2018-06-13:
@ihormelnyk I used the exact schematic and components as in this post (http://ihormelnyk.com/Page/arduino_opentherm_controller) with the exception that I used a rectifier IC that I still had laying around instead of 4 separate diodes. 

Here are two screenshots from Eagle, I used to board layout to visualise how the components fit on the breadboard but I soldered the connections by hand so the traces are different in reality of course.
[image]
[image]

I'll update this post with some pictures of the physical breadboard and the connections when I get home.

ihormelnyk, 2018-06-13:
So you've used this sketch https://github.com/ihormelnyk/opentherm_library/blob/master/examples/OpenTherm_Demo/OpenTherm_Demo.ino and J2.IN is connected  to UNO GPIO2 and J2.OUT to GPIO3? Does boiler enable heating when adapter is connected?

zenz, 2021-11-10:
Any updates for this topic? I have a BAXI boiler which run with OT 3.0. But I found that with This OT Library, I can only read states from boiler, but cannot set any thing including boiler status(heaitng on off extra).

diyless, 2021-11-10:
@zenz there is 4.2 version already, it is fully compatible with 2.2 version, there are no breaking changes.

ihormelnyk, 2021-11-10:
You are not the first who is saying about weird Baxi logic. A common practice in addition to clearing CH flag is to set the boiler temperature to 10 degrees celsius.

pavon87, 2021-12-14:
Hi @senz
Which boiler do you have?  Recently I install a BAXI Compact Platinium 30/30, and almost everything it's working fine (enable / disable hot water, set temperature of hot water, enable / disable heating, setting the temperature of heating with PID) except the most important, tell to boiler when must start heating or stop, in other words, the thermostat.

Evgen2, 2022-02-11:
> Any updates for this topic? I have a BAXI boiler which run with OT 3.0. But I found that with This OT Library, I can only read states from boiler, but cannot set any thing including boiler status(heaitng on off extra).

Just look at code and wonder why setBoilerStatus() call buildSetBoilerStatusRequest()  were in turn 
	return buildRequest(OpenThermMessageType::READ_DATA, OpenThermMessageID::Status, data);

seems that setBoilerStatus() really is getBoilerStatus(), and right setBoilerStatus() must call buildRequest(OpenThermMessageType::WRITE_DATA, ....

zenz, 2022-02-11:
> 

Should read the OT protocol definition. READ_DATA is correct for this MSG ID.

zenz, 2022-02-11:
> > 
> 
> Should read the OT protocol definition. READ_DATA is correct for this MSG ID.

> Just look at code and wonder why setBoilerStatus() call buildSetBoilerStatusRequest() were in turn return buildRequest(OpenThermMessageType::READ_DATA, OpenThermMessageID::Status, data);
>
> seems that [...]

Should read the OT protocol definition. READ_DATA is correct for this MSG ID.

Chupaka, 2022-02-11:
@Evgen2 a quotation from OpenTherm 2.2 specification, just in case:

> The status exchange is a special form of conversation which should be initiated by the master by sending a
READ-DATA(id=0,MasterStatus,00) message. The slave must respond with READ-ACK(id=0,MasterStatus,
SlaveStatus) to send back the Slave Status information in the same single conversation. Since it is
mandatory to support this data object, the slave cannot respond with DATA-INVALID or UNKNOWN-DATAID. A
WRITE-DATA(id=0,...) from the master should not be used.

pavon87, 2022-02-11:
In my case, with a BAXI boiler with OT 3.0 using the default library opentherm_library, almost everything it's working fine  (enable / disable hot water, set temperature of hot water, enable / disable heating, setting the temperature of heating with PID) except tell to boiler when must start heating or stop, in other words, the thermostat. If I enable the heating, it's start heating and if I disable heating it stops, but I think that is not the correct way.....
Any idea about it?

Evgen2, 2022-02-11:
> Should read the OT protocol definition. READ_DATA is correct for this MSG ID.
OT protocol  makes me crazy 

OK

> I can only read states from boiler, but cannot set any thing including boiler status(heaitng on off extra).
but how you fix it?
My Navien  boiler  has jumper  for OT using,  and  with this jumper set OFF it work with OT controller just in read-only mode

zenz, 2022-02-11:
>> but how you fix it? My Navien boiler has jumper for OT using, and with this jumper set OFF it work with OT controller just in read-only mode

I already write down my solution with BAXI boiler, please check my above post.

zenz, 2022-02-11:
> In my case, with a BAXI boiler with OT 3.0 using the default library opentherm_library, almost everything it's working fine (enable / disable hot water, set temperature of hot water, enable / [...]

Check this repo for what you want.
https://github.com/diyless/esp32-wifi-thermostat

Evgen2, 2022-02-12:
> Check this repo for what you want. https://github.com/diyless/esp32-wifi-thermostat

very interesting. 
Some remarks  
1. I see no delay in loop(), so you handle boiler's controller as fast as possible without idle time
2.  Compared to original ihormelnyk's code you  check responce's id in processResponse(). In my case I get ones in some minutes response id=0 (Status) in response to  request with non zero id. Strange thing, but  check responce's id  deal with it. 
3.  You use 1 second interval for PID. Seems it is too fast  for a slow home heating process

zenz, 2022-02-14:
> very interesting. Some remarks
>
>     1. I see no delay in loop(), so you handle boiler's controller as fast as possible without idle time
>
>     2. Compared to original ihormelnyk's code you  check [...]

Please read OT definition.
DIYLESS's code using async method to send request and handle response. it's the very good idea to handle things. 1second is not very fast indeed, especially when you need to handle the fast change CH temp so that the boiler not to enter the protection mode.

zenz, 2022-02-14:
@Evgen2 
You don't need to get CH temp per second(actually even in async mode, you cannot reach that), but you need to communicate with OT slave per second(and also need to guarantee the request was successfully sent to slave, so sometimes you may need to resent the request).
most of the boilers have a temp exceeded protection, such like you set it to 45C, your boiler is a 24kw/h boiler, min output is about 10kW/h, but your room  only need 8kW/h, in this situation, your boiler will reach the set point very quick, but your room indeed not reach the needed temp. then the boiler goes to protect mode without burning for about 3minutes, your room will slowly cold down. 
So a better way is watching the CH temp as fast as possible and change the set point 5C higher than the CH temp every 5 seconds, so that the boiler will not go to protect mode until your room reach the temperature in above situation(we call it low load mode).

Evgen2, 2022-02-14:
> @Evgen2 You don't need to get CH temp per second(actually even in async mode, you cannot reach that), but you need to communicate with OT slave per second(and also need to guarantee the request was successfully sent to slave, so sometimes you may need to resent the request). 

and in  your code there is no limits for number of communications per second

> most of the boilers have a temp exceeded protection, such like you set it to 45C, your boiler is a 24kw/h boiler, min output is about 10kW/h, but your room only need 8kW/h, in this situation, your [...]

In my case  slow down from 65 to 45C output T take about 15 min. From 45 to 65 it take about 20 min, with delay about 1 min from time of command to burning flame, next 30 sec  took low flame with 15% modulation, next 2 min goes to 40% .
So all processes are not so fast

rmull561, 2024-09-17:
I'm trying to switch "eco" mode on a de Dietrich boiler. It controls DHW pre-heating. It can be set in the configuration menu of the boiler, but I don't know how to set it in OpenTherm. Their OpenTherm-based thermostat AD301 has this feature documented in the manual, so I think it should be possible.

The boiler reports OT version 3.0. I tried setting the bit 6 in the master status, but that disables DHW completely. I tried also setting the bit 7 and all bits in the master configuration, but it doesn't seem to be doing anything. There are zero TSPs reported and no other remote parameters than the two specified in OT2.2. I tried iterating over all unspecified IDs and found four that can be read: 35, 36, 113, 114. The first two change value when the boiler is heating, one of them looks like fan speed. The other two always return zero.

Any suggestions what else could I try?

rmull561, 2024-09-18:
I'm dumb. Only now I tried the obvious thing. It's the bit 1 of the master status which disables and enables the eco mode on this boiler. I guess that's the reason why bit 6 was added later to have something to actually disable DHW. What a mess of specification.