ihormelnyk/opentherm_library #2: strange return value

geo903, 2018-09-29:
Hello! Thank you so much for your development!
I use an example of your code to get the value of the boiler modulation and get a strange value, if it is divided by 256 then do not get a number from 0 to 100%

```
  unsigned int data = 0xFFFF;
  unsigned long request = ot.buildRequest(
  OpenThermRequestType::READ,
  OpenThermMessageID::RelModLevel,
  data);
  Serial.println("mod:"+String(ot.sendRequest(request)));
```

```
Central Heating: on
Hot Water: off
Flame: off
diagnost: 0
fault: 0
**mod:3222339584**
Boiler temperature is 36.80 degrees C
```

ihormelnyk, 2018-09-29:
Hi @geo903 ,
To get modulation level you have to parse response like this:
float modulation_level = (response & 0xFFFF) / 256.0;

geo903, 2018-09-30:
Thank you! it works!

closed 2018-10-22 by ihormelnyk