ihormelnyk/opentherm_library #11: Optimising script speed?

Tommrodrigues, 2019-02-16:
Hi there,

I've had a lot of fun using your OpenTherm project, so much so that I've attempted to implement it in a project for Homebridge integration via a NodeMCU which works almost perfectly (you can find the project [here](https://github.com/Tommrodrigues/homebridge-nodemcu-thermostat). However, when implementing with Homebridge, response time is extremely important. One problem I have noticed is that when the NodeMCU is communicating with the boiler, this takes a while and means that HTTP requests are unable to be handled. Here is a small example snipped of code I have made up to illustrate my point:

```
void loop() {
  if (new_ts - ts > 1000) {
    bool enableCentralHeating = true;
    bool enableHotWater = true;
    bool enableCooling = false;
    unsigned long response = ot.setBoilerStatus(enableCentralHeating, enableHotWater, enableCooling);
    OpenThermResponseStatus responseStatus = ot.getLastResponseStatus();

    if (responseStatus != OpenThermResponseStatus::SUCCESS) {
      Serial.println("Error: Invalid boiler response " + String(response, HEX));
    } else {
      ot.setBoilerTemperature(boilerTargetTemp);
    }
  }

  if (request.indexOf("/targetTemperature") != -1)  {
     tempString = "";
     tempString = (request.substring(23,25));
     tempString.trim();
     boilerTargetTemp = tempString.toFloat();
  }
}
```

As you can see, every 1s, the boiler is contacted to set the statuses as well as the temperature. Also note that on **every** repeat, the NodeMCU checks to see if `/targetTemperature` has been requested via HTTP. If it has, it sets the boiler temperature variable to the desired value included in the HTTP request.

Now, whilst the NodeMCU has not reached a full second (when it would then need to communicate with the boiler), it will simply check if an HTTP request has been made, skipping the `if (new_ts - ts > 1000) {` boiler control section. At this point, the NodeMCU responds straight away with no problems. However, the problem comes when an HTTP request is made whilst the boiler control function is in progress. At this point, the HTTP request is put on hold and is only responded to once the boiler control function has finished (which can take a while and make Homebridge unhappy).

Thus, would it be possible to remove or enhance some of the boiler control function to reduce this 'hanging' time? For instance, could the code be simplified down to:

```
void loop() {
  if (new_ts - ts > 1000) {
    bool enableCentralHeating = true;
    bool enableHotWater = true;
    bool enableCooling = false;
    unsigned long response = ot.setBoilerStatus(enableCentralHeating, enableHotWater, enableCooling);
   ot.setBoilerTemperature(boilerTargetTemp);
    }

  if (request.indexOf("/targetTemperature") != -1)  {
     tempString = "";
     tempString = (request.substring(23,25));
     tempString.trim();
     boilerTargetTemp = tempString.toFloat();
  }
}
```

I.e. removing the reading of a response status? My question is, will removing these lines of code reduce the amount of time spent in this section of the code? I am asking because I am unaware at exactly which point the status is read. For example, after running `unsigned long response = ot.setBoilerStatus(enableCentralHeating, enableHotWater, enableCooling);` does the code automatically wait 1s for a response or does this only happen when I explicitly specify `if (responseStatus != OpenThermResponseStatus::SUCCESS) {`? 

Thank you in advance for your help

ihormelnyk, 2019-02-16:
Hi @Tommrodrigues ,
Status checking is very quick operation and removing it won't help.
According to protocol documentation you have to send any command but at least every second.
So to reduce loop time you can send one command per sec, just switch between status and setTemperature commands.

Also if you are not aware of response, you can send requests asynchronously, then sending request will take only 34ms.

How to send async request please read here http://ihormelnyk.com/opentherm_library

Tommrodrigues, 2019-02-16:
Great, thank you for the information,  I'll have a look into async requests.

Thank you :)

closed 2019-02-16 by Tommrodrigues