ihormelnyk/opentherm_library #13: Asynchronous requests example?

Tommrodrigues, 2019-04-20:
Hi there,

I'm currently developing a script [here](https://github.com/Tommrodrigues/homebridge-nodemcu-boiler/blob/master/NodeMCU-Boiler-TMP36.ino) which aims to integrate OpenTherm with [Homebridge](https://github.com/nfarina/homebridge).

In this script, I attempt to make quite a few requests at the same time and have noticed that there is a large delay in doing so, causing the script to run slowly. After doing some looking around, I saw that you mentioned Asynchronous requests on [your website](http://ihormelnyk.com/opentherm_library) and assume that these will be able to fix the issue by implementing them on all of the requests I make to OpenTherm. 

However, whilst you did include instructions on how to implement these requests, I am still struggling to understand exactly how to implement them. Therefore, would you be able to give a short example showing how to implement asynchronous requests?

To help illustrate my point, here is a simplified version of my script:

```c++
#include <OpenTherm.h>

const int inPin = 4;
const int outPin = 5;
OpenTherm ot(inPin, outPin);

unsigned long request, response;
float boilerTempCurrent, boilerTargetTemp, dhwTemp;
int currentHeatingCoolingState, domesticHotWater;
bool enableCentralHeating = true;
bool enableHotWater = true;
bool enableCooling = false;
float dhwSetPoint = 50;

void handleInterrupt() {
  ot.handleInterrupt();
}

void setup() {
  ot.begin(handleInterrupt);
}

void loop() {
  //Get CH temperature
  boilerTempCurrent = ot.getBoilerTemperature();

  //Set statuses
  response = ot.setBoilerStatus(enableCentralHeating, enableHotWater, enableCooling);
  OpenThermResponseStatus responseStatus = ot.getLastResponseStatus();
  if (responseStatus != OpenThermResponseStatus::SUCCESS) {
    Serial.println("Error: Invalid boiler response " + String(response, HEX));
  } else {

    //Set CH temperature
    ot.setBoilerTemperature(boilerTargetTemp);

    //Get CH state
    currentHeatingCoolingState = ot.isCentralHeatingEnabled(response);

    //Get DHW state
    domesticHotWater = ot.isHotWaterEnabled(response);

    // Set DHW temperature
    unsigned int data = ot.temperatureToData(dhwSetPoint);
    request = ot.buildRequest(
      OpenThermRequestType::WRITE,
      OpenThermMessageID::TdhwSet,
      data
    );
    ot.sendRequest(request);

    // Get DHW temperature
    request = ot.buildRequest(
      OpenThermRequestType::READ,
      OpenThermMessageID::Tdhw,
      0x0000
    );
    response = ot.sendRequest(request);
    dhwTemp = ot.getTemperature(response);
  }
  //Simulate `milis()`
  delay(1000);
}

```

Thank you in advance for your help,

Kind regards, Tom

miksumin, 2021-03-15:
Look here - https://github.com/miksumin/opentherm_library/tree/master/examples/OpenThermMaster_Async

ProphetOfDoom, 2024-01-31:
I know it's been a while since the above post, but I thought I'd add my recent observations.

I flashed the following Arduino sketch by the same author as above to my ESP32-DevkitC-V4 board with DiyLess ESP32/ESP8266 Thermostat Shield, connected to my (nearly new) Baxi 824 System Boiler.

https://github.com/miksumin/OpenTherm/tree/master/examples/OpenThermTestIDs

This produced the attached output, but worringly, left the boiler flashing an E83 (Communication Error) message, which repeatedly returned after disconnecting the ESP32 kit and power cycling the boiler to return to Hive 230v control. I couldn't find anything online on how to clear this error message, but eventually reasoned that it might have been due to the final request/response in the above sketch returning an error, as shown in the log. So, I then flashed the ESP32 with a simple sketch that periodically just gets the boiler temperature as shown below, and to my great relief, this cleared the boiler error message.

```
  // Get Boiler Temperature
  ch_temperature = ot.getBoilerTemperature();
  Serial.println("CH temperature is " + String(ch_temperature) + " degrees C");
```
Looking at the output log, it seems that my Baxi 824 System Boiler is OpenTherm 4.0, information I surprisingly couldn't find in any of the manuals or online!

Hope this helps someone.

[arduino_output.log](https://github.com/ihormelnyk/opentherm_library/files/14117487/arduino_output.log)