ihormelnyk/opentherm_library #17: ISR not in IRAM issue ESP8266

Tommrodrigues, 2019-05-19:
Hi there,

I'm having problems using the library with the newest ESP8266 release (`v2.5.1`). Specifically, when trying to call:

```c++
void handleInterrupt() {
  ot.handleInterrupt();
}
```

It seems others have been having this issue in similar situations as you can see [here](https://github.com/esp8266/Arduino/issues/6087#issuecomment-492012766). I get a very similar error message to the one I linked - something like: 

```
ISR not in IRAM!

Abort called

>>>stack>>>
```

As I understand it, the correct way to solve this issue is to change the library to move the ISR into IRAM, otherwise the script never proceeds past this interrupt function. However, I'm unsure how you would do this otherwise I would've made a pull request to resolve the issue.

Do you have any ideas?

Kind regards, 
Tom

P.S. I am using ESP8266 `v2.5.0` for now as the issue is only with the newest version

verner-lall, 2019-05-26:
Add ICACHE_RAM_ATTR to your interrupt handler.

`void ICACHE_RAM_ATTR handleInterrupt() { 
    //handle
}
`

Tommrodrigues, 2019-05-26:
@verner-lall Thank you, that does seem to fix the problem! However it looks like a bit of a bodge - as you say, I think the better solution might bee to add it to the library's interrupt handler, just not quite sure where I'd do that. Got any ideas?