ihormelnyk/opentherm_library #24: Boot loop

mkrawcz1, 2020-05-01:
Hi :-)
My ESP8266s keep rebooting after "OpenThermMaster_Demo.ino" upload with message:
"
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
"
My environment:
- ESP-12E (different modules),
- Arduino v. 1.8.12, ESP8266 v. 2.6.3, 
- OpenTherm v. 1.1.0,
- inPin = 4,
- outPin = 5.

Please help :-)
Regards
MK

ihormelnyk, 2020-05-01:
Hi @mkrawcz1,
Try to use ICACHE_RAM_ATTR attribute:
void ICACHE_RAM_ATTR handleInterrupt(

mkrawcz1, 2020-05-01:
It seems that it solved the problem :-)

mkrawcz1, 2020-05-03:
Hi
Another issue... I'm trying to interface with De Dietrich MCA 35 Innovens, which claims it's compatible with OpenTherm. The issue is that after a few successfull answers from boiler I'm experiencing no answers for following messages. I'm using template sketch "OpenThermMaster_Demo.ino". Anyone had success witch interfacing this boiler?

closed 2020-05-09 by mkrawcz1