ihormelnyk/opentherm_library #27: Can't build in Platformio, getTemperature is missing in cpp

ZzubbuzZ, 2020-11-12:
I download the master version of your lib to my project (I'am using platformio 5.0.1), I'm just updating an existing project that used OpenTherm::getTemperature().
The signature OpenTherm::getTemperature is present in your headers (l153), but not in your cpp...??

I fix it by adding this :
```c++
float OpenTherm::getTemperature(unsigned long response) {
	float temperature = isValidResponse(response) ? getFloat(response) : 0;
	return temperature;
}
```

warnerthuis, 2020-11-29:
Is a fix coming for this error?

ihormelnyk, 2020-11-29:
getTemperature is obsolete and should be removed
use getFloat function to parse response

Chupaka, 2021-04-01:
https://github.com/ihormelnyk/opentherm_library/commit/908d43d7048ab51e003a40d195959a8624942a20 fixed this, so the issue should be closed, I believe?