ihormelnyk/opentherm_library #29: Viessmann Vitodens 100

pruwait, 2021-01-24:
Спасибо за полезную библиотеку.
Вижу, что в Тасмота для реализации OpenTherm используется именно она.
У меня возникло несколько вопросов по использованию библиотеки.
1. В момент опроса котла Viessmann по протоколу OPENTHERM перестаёт работать управление котлом через копки на его панели. Можно ли это исправить каким-то иным режимом?
2. При подключении к котлу сразу задаются установки теплоносителя. Можно ли сначала считать имеющиеся параметры, а затем, при необходимости их менять? Возможно, это вопрос по настройке Тасмота, но всё же надеюсь на помощь автора и сообщества, которому интересна работа по OpenTherm.
3. При отключении OpenTherm котел реализует погодозависимую теплогенерацию. С панели задаю температуру ГВС и параметр кривой отопления, от которого зависит текущая температура теплоносителя с учетом наружной температуры. Наружная текущая температура также отражается на дисплее котла. 
3.1. Возможно ли получить по OpenTherm текущую наружную температуру?
3.2. Возможно ли считать и менять через OpenTherm параметр кривой отопления по которому меняется погодозависимая теплогенерация?

ihormelnyk, 2021-01-24:
1. Думаю нельзя, уплавление или с панели или с термостата.
2. Команда установки темпрературы теплоносителя не имеет режима чтения, есть только команда чтения текущей термпературы теплоносителя.
3.1 Есть команда чтения наружной температуры 
3.2 Наверное нет, да и нет смысла менять параметр кривой, так как теперь управление теплоносителем ложитса на термостат и тогда он должен строить эту кривую и по ней задавать темпрературу.

С помощью библиотеки вы можете отправлять любые команды описанные в протоколе, но не все подерживаються вашим котлом.
Ссылка на документацию по протоколу есть в статье
http://ihormelnyk.com/opentherm_library

pruwait, 2021-06-29:
Все вопросы были закрыты. Наружную температуру Тасмота отдаёт по MQTT в hassio, алгоритм теплогенерации прописан как автоматизация в hassio. Работает с января 2021 по текущий момент.

closed 2021-06-29 by pruwait