ihormelnyk/opentherm_library #31: [question] Async response handler: sending back responses doesn't work without delay

robertklep, 2021-04-13:
I have an issue with sending back responses from an async response handler. My code looks like this:
```
void ICACHE_RAM_ATTR handleInterrupt() {
  ot.handleInterrupt();
}

void processRequest(unsigned long request, OpenThermResponseStatus status) {
  if (status != OpenThermResponseStatus::SUCCESS) return;

  auto dataId = ot.getDataID(request);
  auto response;

  if (dataId == 0) {
    response = ot.buildResponse(…);
  } else if (dataId == 1) {
    response = ot.buildResponse(…);
  } else { … }

  ot.sendResponse(response);
}

void setup() {
  ot.begin(handleInterrupt, processRequest);
}

void loop() {
  ot.process();
}
```

Basically, I'm trying to build a (very minimal) boiler simulator by responding to commands sent by a thermostat.

The issue is that unless I add a delay (of about 200ms) before `ot.sendResponse()`, the response seems to put the thermostat in an error mode. Do I have to wait for a specific event before I can send back a response? Or am I doing something wrong entirely?

For what it's worth, the thermostat is sending quite a lot of messages, mostly READ_DATA for different types of data ID's.

ihormelnyk, 2021-04-13:
Hi @robertklep ,
It is necessary to add a delay before a response.
According to OT protocol specification:

> The slave is expected to respond with a single frame reply within a defined period of 20ms to 800ms from the end of the master transmission.
> The master unit must wait 100ms from the end of a previous conversation before initiating a new
> conversation. The master must communicate at least every 1 sec

robertklep, 2021-04-13:
@ihormelnyk thanks! I changed the delay to 30ms to be safe and that seems to be acceptable 👍🏻

closed 2021-04-13 by robertklep