ihormelnyk/opentherm_library #32: ISR not in IRAM!

akamming, 2021-05-10:
Hi,

great work here. since i want to connect my boiler to my domoticz installation, i purchased your opentherm adapter and shield. 

My approach is to get your [opentherm thermostat](http://ihormelnyk.com/opentherm_thermostat)  sample up and running and then change it so i can manage the boiler temperature from my domoticz instance.

so i followed the steps to get the example up and running, it fails to start. It just starts rebooting all over again on an ISR not in Iram exception, with the  following loglines endlessly repeated in the serial monitor:

```
........
Connected to <mySSiD>
IP address: <MyIP>
MDNS responder started
HTTP server started
ISR not in IRAM!

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffe90 end: 3fffffc0 offset: 0000
3ffffe90:  00000005 00000040 00000020 00000000  
3ffffea0:  000000fe 00000000 00000000 00000000  
3ffffeb0:  00000000 00000000 00000000 00ff0000  
3ffffec0:  5ffffe00 5ffffe00 3ffefa64 00000000  
3ffffed0:  00000003 00000004 3ffee444 4020d1f2  
3ffffee0:  4010076e 00000000 4bc6a7f0 4020d204  
3ffffef0:  00000000 4bc6a7f0 3ffee444 4020d711  
3fffff00:  00000000 00000000 4bc6a7f0 00000000  
3fffff10:  3fffff40 3ffee46c 40100504 3ffefa30  
3fffff20:  00000000 402010c0 3ffee444 4020d7c0  
3fffff30:  0019e0aa 3ffee4a0 3ffee46c 4020ac5d  
3fffff40:  00000001 3ffee4a0 3ffee46c 4020ac88  
3fffff50:  00000001 3ffee4a0 3ffee46c 40201ded  
3fffff60:  402035e0 0502a8c0 4020e7d4 4020e7c0  
3fffff70:  4020ef48 6d657400 feef0070 80efeffe  
3fffff80:  3ffefa30 00000000 feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffee6c8  
3fffffa0:  3fffdad0 00000000 3ffee688 4020cbc0  
3fffffb0:  feefeffe feefeffe 3ffe84f0 40101069  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

```

I did some tracing, it fails the statement "ot.begin(handleInterrupt);"

any idea how to make the sample work?

ihormelnyk, 2021-05-10:
Hi @akamming ,
Just add ICACHE_RAM_ATTR attribute to handleInterrupt interrupt handler.
Like in sample here
https://github.com/ihormelnyk/opentherm_library/blob/master/examples/OpenThermMaster_Demo/OpenThermMaster_Demo.ino

akamming, 2021-05-10:
NIce.. that did the trick.. tx! 

might not be a bad idea the sample code on that page as well! other might run into this...

anyway tx a lot and keep the good work! (now getting "invalid boiler response 0", which is expected, since the adapter was just ordered and not deliverd...)

closed 2021-05-10 by akamming