ihormelnyk/opentherm_library #33: [question] Data timing

robertklep, 2021-05-17:
Hi!

I was reading through the library code, trying to comprehend the timing it applies, specifically the usage of 750μs [here](https://github.com/ihormelnyk/opentherm_library/blob/7baa76a5ae471fcb3a6d55775b5aa65c823bb2e1/src/OpenTherm.cpp#L149) and [here](https://github.com/ihormelnyk/opentherm_library/blob/7baa76a5ae471fcb3a6d55775b5aa65c823bb2e1/src/OpenTherm.cpp#L160).

If I understand the code and the OpenTherm specification correctly, shouldn't that be 650μs (500μs nominal plus an additional 150μs for the maximum transition period)?

[image]

Also, strictly speaking, shouldn't a timeout occur after 1150μs instead of 1000μs as it does now?

Reason I'm asking is because I'm trying to track down some issues with invalid messages being received from my thermostat. I'll try and see if changing the timing fixes them, but just wondering if I understand the code correctly 😊

ihormelnyk, 2021-05-17:
750μs - 500μs+250μs - to check 1->0, 0->1 bit transition
1000μs timeout is ok, because there should be interrupt every half bit (500μs)

robertklep, 2021-05-18:
@ihormelnyk thanks!

I still don't understand where the 250μs is coming from. I assume it's based on the 100+150μs transition period, but from how I understand the specification, the 100μs is how much _shorter_ than 500μs a transition may be. So a valid half bit period is 400μs to 650μs, isn't that right?

IgorYbema, 2021-10-01:
So I was looking at it also just to understand it.
750ms is just the half bit time (500ms) plus another 250ms (quarter of the bit time). This allows it to sync with the mid-bit transistions (synced at the first start bit) and ignore the begin-of-bit transitions (if machester encoding requires a transitions at begin of the bit). As you state, half bits are in between 400 and 650ms. So another transistion (triggered by the interrupt gpio change input) with is at least 750ms must always be the transition for the mid-bit point (counted from the first start bit).
So the code essentiantly ignores the begin-of-bit transitions and only read the mid-bit transitions.