ihormelnyk/opentherm_library #40: Heating start / stop (no enable /disable)

pavon87, 2021-11-20:
Hi 
First of all, thanks for the great job with this library, it's really nice have an open source alternative to communicate with the boilers.

Recently I install a BAXI-ROCA boiler, compatible with opentherm and I bought the opentherm adapter with shield for ESP8266. From the examples available, I created a fork with MQTT to integrate it to Home Assistant and use as temperature reference different sensors. 

Almost everything it's working fine (enable / disable hot water, set temperature of hot water, enable / disable heating, setting the temperature of heating with PID) except the most important, tell to boiler when must start heating or stop, in other words, the thermostat.  If I enable the heating, it's start heating and if I disable heating it stops, but I think that is not the correct way.....

Without the opentherm connected I can use a relay to start or stop the heating keeping the heating enabled, so I know that it's something that I miss in the code, but I cant' find it in the library definition.

any help will be welcome ;)

Thanks in advance

shariecas, 2022-05-03:
Hello,

I ran into exact same problem. My boiler is Baxi Prime HT 1.240. According to the manual, it is able to work with Siemens QAA73 room thermostat, which uses OpenTherm.
I have connected an Arduino using Ihor Melnyk's adapter. I can read the status (cmd ID 0), configuration (cmd ID 3), fault flags (cmd ID 5), relative modulation level (cmd ID 17), boiler temperature (cmd ID 25), slave OpenTherm version (cmd ID 125, for some reason it replies "9.02") and slave version (cmd ID 127).
I can write control setpoint (cmd ID 1), maximum relative modulation level (cmd ID 14) and cmd IDs 124 and 126. Strangely, I cannot read the outside temperature, since it is supported on QAA73. And the main problem, the boiler wont start, although control setpoint is higher than boiler temperature. 

What am I missing?

Best regards.

littlej956, 2023-10-12:
> I got it working. After reading slave configuration (cmd ID 3), the MemberID code is received in the response. After that, master configuration write command (cmd ID 2) needs to be sent with the same MemberID code. It even states in the OpenTherm 2.2 specification that "A valid Read Slave Configuration and Write Master Configuration message exchange is recommended before control and status information is transmitted."
> 

Can you post a code example or something ? I have same issue with a viessman vitodens 100-w :(

Llenas09, 2023-11-20:
@pavon87 Hi can you share your code?. I'm on the same situation I get a Baxi Boiler to.

littlej956, 2023-11-20:
@Llenas09 check my mentions, this is the code snippet you need 

    //read slave config
    unsigned int data = 0;
    unsigned long request = ot.buildRequest(
        OpenThermRequestType::READ,
        OpenThermMessageID::SConfigSMemberIDcode,
        data);
    unsigned long response = ot.sendRequest(request);
    data = ot.getUInt(response);
    data &= 0xFF; //clear high byte
    Serial.println("Boiler Member Id: " + String(data));

    //write the same data as master id
    request = ot.buildRequest(
        OpenThermRequestType::WRITE,
        OpenThermMessageID::MConfigMMemberIDcode,
        data);
    response = ot.sendRequest(request);

Llenas09, 2023-11-20:
Thanks for the info @littlej956 but i'm interested on the MQTT part to. This is why I asked for all the code from @pavon87. Thanks for the help!

Baxi has a mode that you can connect an outdoor temp sensor and respond to extra demand if the temp is very low. Any try with this?

Llenas09, 2023-11-20:
Thanks for the info @littlej956 but i'm interested on the MQTT part to. This is why I asked for all the code from @pavon87. Thanks for the help!

Baxi has a mode that you can connect an outdoor temp sensor and respond to extra demand if the temp is very low.

pavon87, 2023-11-20:
Hello @Llenas09 
I pushed my last version to GitHub and set it to public. It is not a perfect code, it is just a fork of my own code that I use with the ESP8266/32 normally with the different types of sensors.

https://github.com/pavon87/opentherm-mqtt

I hope that it will be useful. 
KR