ihormelnyk/opentherm_library #41: Cannot communicate with Intergas Kompakt HR 22

rcambrj, 2022-01-30:
I have an [Intergas Kompakt HR 22](https://www.intergas-verwarming.nl/zakelijk/producten/kombi-kompakt-hr/hr-22/) and normally it's connected to a Honeywell T87M OpenTherm thermostat.

I've put [this basic OpenTherm sketch](https://github.com/ihormelnyk/opentherm_thermostat) onto an ESP8266 (Wemos D1 mini) which has a [DIYLESS Master OpenTherm Shield](https://diyless.com/product/master-opentherm-shield) on it, and connected it to the boiler instead of the Honeywell thermostat, and all I get is:

```
Error: Invalid boiler response 0
```

Pins are correctly configured:
```
const int inPin = 4; // D2
const int outPin = 5; // D1
```

I'm not entirely sure how to move forward with debugging, any help would be appreciated 😢

craigsbits, 2023-02-08:
Possibly unable to assist but I am about to try the same configuration (waiting for new boiler).  Just to check is the web http page giving sensible response - looks ok!  Might be worth swapping ot1 & ot2 wires I know it should not matter but....

craigsbits, 2023-02-08:
Oh flip I am a whole year out from everyone else!!!

wjaszczuk, 2023-02-19:
@craigsbits it seems I'm facing the same problem here. I'm trying to use a gateway between my boiler (Immergas Vitrix II Zeus ErP) and the official thermostat - CAR2.
I'm using WEMOS D1 mini ESP8266 and I always get an Invalid response there.
The voltage between OT wires is something about 6V. When I use CAR2  there is 24-25V.

I tried to swap OT wires - it didn't help.