ihormelnyk/opentherm_library #48: Only getting invalid response on Master side

ameerhazo, 2023-01-19:
Hello @ihormelnyk an everyone, thank you for providing this library to use for free. I'm trying to use the Slave example as a way to simulate a boiler. The Slave is receiving the requests from the Master without any problems. Unfortunately, the responses I have sent out from the Slave back to the Master has always been considered as invalid responses by the Master. Debugging the response frame showed me that I'm only receiving 0s from the Slave.  I would appreciate it if anyone who has had this problem to be able to give me tips on what they did to fix the problem. Thank you and kind regards.

Ameer

ihormelnyk, 2023-01-19:
Hi @ameerhazo,
try to add a delay before sending the response
delay(20); //20..400ms, usually 100ms
ot.sendResponse(response);

ameerhazo, 2023-01-19:
Hello @ihormelnyk, it worked! Thank you so much for being such a lifesaver :D

closed 2023-01-19 by ameerhazo