ihormelnyk/opentherm_library #53: Help needed - boiler not starting - viessman vitodens 100-w

littlej956, 2023-10-13:
Hi, i have a viessmann vitodens 100-w with underfloor heating, i have successfully connected via opentherm to the boiler, i ca set direct hot water temperature and boiler temperature and successfully updates on the boiler lcd screen.

I have a few problems i cannot seems to manage and any help will be appreciated (bare with me, i'm a new to codding) :

1. changing heating and water temperature works but looks like the boiler is not starting, no flame and no pump running, i have tried to send the commands mentions in this issue https://github.com/ihormelnyk/opentherm_library/issues/40 but i always get an error 40000302 (not sure how to decode this)

2. from time to time, my boiler starts and ramps temperature to ~80C and waits to cool, i have no idea why this is happening since hot water is set to 45c and boiler is set to 35C.

3. is there a way to tell the boiler max modulation to heat the water? eg like ~20% ? i want to stay as low as possible since i don't need high temp with underfloor heating

Thanks again for any help/explanation, your time is appreciated.

ihormelnyk, 2023-10-13:
Hi @littlej956,
Maybe in your case, the boiler manufacturer added such behavior that the boiler accepts commands but does not perform some actions if thermostat member id is not equal to the manufacturer member id.
So you can try to read slave configuration first to get member id and then send master configuration with the same member id.

To limit temperature it should be enough just to set some temperature manually on the boiler after that OT protocol will be able to set temperature only below that limit.

Below you can find master sample sketch with added reading/writing master/slave configuration, try to use it

```
#include <Arduino.h>
#include <OpenTherm.h>

const int inPin = 21;  //for Arduino, 4 for ESP8266 (D2), 21 for ESP32
const int outPin = 22; //for Arduino, 5 for ESP8266 (D1), 22 for ESP32
OpenTherm ot(inPin, outPin);

void IRAM_ATTR handleInterrupt() {
    ot.handleInterrupt();
}

void setup()
{
    Serial.begin(9600);
    Serial.println("Start");
    ot.begin(handleInterrupt);
}

void loop()
{
    //read slave config
    unsigned int data = 0;
    unsigned long request = ot.buildRequest(
        OpenThermRequestType::READ,
        OpenThermMessageID::SConfigSMemberIDcode,
        data);
    unsigned long response = ot.sendRequest(request);
    data = ot.getUInt(response);
    data &= 0xFF; //clear high byte
    Serial.println("Boiler Member Id: " + String(data));

    //write the same data as master id
    request = ot.buildRequest(
        OpenThermRequestType::WRITE,
        OpenThermMessageID::MConfigMMemberIDcode,
        data);
    response = ot.sendRequest(request);

    //Set/Get Boiler Status
    bool enableCentralHeating = true;
    bool enableHotWater = true;
    bool enableCooling = false;
    response = ot.setBoilerStatus(enableCentralHeating, enableHotWater, enableCooling);
    OpenThermResponseStatus responseStatus = ot.getLastResponseStatus();
    if (responseStatus == OpenThermResponseStatus::SUCCESS) {
        Serial.println("Central Heating: " + String(ot.isCentralHeatingActive(response) ? "on" : "off"));
        Serial.println("Hot Water: " + String(ot.isHotWaterActive(response) ? "on" : "off"));
        Serial.println("Flame: " + String(ot.isFlameOn(response) ? "on" : "off"));
    }
    if (responseStatus == OpenThermResponseStatus::NONE) {
        Serial.println("Error: OpenTherm is not initialized");
    }
    else if (responseStatus == OpenThermResponseStatus::INVALID) {
        Serial.println("Error: Invalid response " + String(response, HEX));
    }
    else if (responseStatus == OpenThermResponseStatus::TIMEOUT) {
        Serial.println("Error: Response timeout");
    }

    //Set Boiler Temperature to 64 degrees C
    ot.setBoilerTemperature(64);

    //Get Boiler Temperature
    float ch_temperature = ot.getBoilerTemperature();
    Serial.println("CH temperature is " + String(ch_temperature) + " degrees C");

    //Set DHW setpoint to 40 degrees C
    ot.setDHWSetpoint(40);

    //Get DHW Temperature
    float dhw_temperature = ot.getDHWTemperature();
    Serial.println("DHW temperature is " + String(dhw_temperature) + " degrees C");

    Serial.println();
    delay(1000);
}
```

littlej956, 2023-10-14:
Hi @ihormelnyk, 

You are absolutely correct thank you for the example, this saved me.  Setting master configuration with the same member id worked and now the boiler turns on and the pump too. 

I can control heating request using **setBoilerTemperature** but unfortunately i cannot turn it off without setting **ot.setBoilerStatus(false, true, false);**, i have tried to **setBoilerTemperature** to 0 but minimum allowed is 20 and pump and flame remains to on. 

Is there something that i'm missing? how should i turn heating request off? 

Thanks again for your time!

ihormelnyk, 2023-10-15:
Hi @littlej956 ,
For most boilers work setting enableCentralHeating to False using the status command and setBoilerTemperature to 10

```
ot.setBoilerStatus(false, true, false);
ot.setBoilerTemperature(10);
```

closed 2023-10-23 by littlej956