ihormelnyk/opentherm_library #54: Not compiling when "thingProperties.h" is used.

vdv-k, 2023-11-04:
When using 
#include "thingProperties.h"
#include <OpenTherm.h> 
an error is generated when compiling the code.
There seems to be a redeclaration of '<unnamed enum> READ' (already used in ThingProperties).

Probably easy to solve but I'm new to all this and don't know how. Any suggestions welcome.

Laxilef, 2023-12-20:
Hey
See this PR: https://github.com/ihormelnyk/opentherm_library/pull/57

4310benno, 2024-02-02:
> When using #include "thingProperties.h" #include <OpenTherm.h> an error is generated when compiling the code. There seems to be a redeclaration of ' READ' (already used in ThingProperties).
> 
> Probably easy to solve but I'm new to all this and don't know how. Any suggestions welcome.

4310benno, 2024-02-02:
Same problem for me in iotCloud : caused by Read definition in Opentherm.h made for "backards compatability".
Solution for the time can be : comment out Read and Write declaration, and change READ in the CPP file to READ_DATA, and upload the library to IOT cloud in a zip

Laxilef, 2024-02-02:
@4310benno did you use the version from the master branch?

4310benno, 2024-02-02:
Yes, and for me it is working now.
Only waiting for your shield to arive😀

Benno

Verstuurd vanaf mijn iPad