ihormelnyk/opentherm_library #60: OpenTherm monitor

ProphetOfDoom, 2024-01-12:
I've got a Baxi 824 System Boiler with Weather Compensation and a Hive Thermostat which is not OpenTherm capable. As my boiler is in my loft, I'd like to be able to view boiler status information such as flow and return temps, weather compensation temp, boiler on/off status etc. I don't plan on changing the Hive Thermostat for the time being. I've ordered a couple of the Master OpenTherm Shields to go with a few ESP32s I've got. 
Do you know if I can just use the OpenTherm shield as a monitor and continue to control the boiler via the (230v) Hive controls, or will using the OpenTherm Shield take complete control of the boiler?

ihormelnyk, 2024-01-12:
Anything connected to the OT terminal takes complete control of the boiler

ProphetOfDoom, 2024-01-14:
Thanks for the prompt reply. Does this mean the boiler then completely ignores the Weather Compensation (IFOS) device?

If this is the case, then I assume that the IFOS Outdoor Temperature value must be available in a reply to the Thermostat, so that the user can choose which Compensation Curve they wish to use, and the Thermostat can then control the setpoint for the Central Heating Flow temp appropriately, and also override it when a call for HW is made.

I'll probably figure all this out for myself in the end, but any info you have will be much appreciated

ihormelnyk, 2024-01-16:
Hi @ProphetOfDoom ,
Yes, you can read the outside temperature and other boiler states using OT protocol.
You can find the link to the OT protocol specification in this post
https://ihormelnyk.com/opentherm_library
There is a list of available commands.

ProphetOfDoom, 2024-01-16:
Thanks for that. I've bookmarked the web site and I'll add the opentherm library to my Arduino install when I get the Master shields I've ordered

ProphetOfDoom, 2024-01-16:
Closed as complete