ihormelnyk/opentherm_library #67: [HELP] Boilers goes berserk when communication with opentherm is lost

littlej956, 2024-06-16:
Hello, i have opentherm working with a Viessman Vitodens 100w for some time now (like 1 year)

My issue, and it might be because of boiler configuration but i have spent some time checking the manual and i cannot find anything relevant about this issue, so i'm here because maybe someone can pinpoint me to a direction.  

Basically it happened that my esp got stuck (i have some watchdog on it) and somehow communication with boiler was lost, losing the communication my boiler goes berserk and heats the water to 95C. (boiler water). I have went over the viessman manual and i cannot find some limit for boiler hot water. 

I have tried to setup via opentherm MaxTSet , but i might get the code wrong since it's not working

**57 R W MaxTSet f8.8 Max CH water setpoint (°C) (Remote parameters 2)**

`    unsigned int tempTemp = ot.temperatureToData(35);
    request = ot.buildRequest(
        OpenThermRequestType::WRITE_DATA,
        OpenThermMessageID::MaxTSet,
        tempTemp);`
        
I plug the power from esp and after ~60 seconds the boiler shows error F.95 (i cannot find it in manual, but on the viessman app it stats communication with thermostat lost). 

So, is there a way to set max temperature for the boiler even if the communication is lost? I'm afraid that will happen at one point when i'm not home or something and will overheat like crazy. 

If anyone has any idea, what to test, what to setup please fell free to add. 

Thanks

Laxilef, 2024-06-21:
On some boilers you can set the maximum and minimum heating temperatures in the service menu.

littlej956, 2024-06-22:
Hello, unfortunately when boiler is configured to work with opentherm thermostat you cannot set service temperature limits. I need to set it up from opentherm somehow.

Laxilef, 2024-06-23:
Hello, from my experience, the boiler should not do anything if communication with the opentherm master is lost. If this works differently on your boiler, I recommend studying the documentation for it in detail.
Most likely, any settings you have set via opentherm will become irrelevant when the master device is disconnected. The boiler will be with the settings that were set without the opentherm device.

Andreotti100, 2024-10-28:
If the protocol is working, in idle the output PIN is high. So, when you're program stalls, the output stays high.
My heater switches back to ON/OFF when there's no communication. Like a relay thermostat.
PIN high: heater full pull, PIN low: no heating.

So, make sure that the output PIN is low when you detect UN-normal behavior.
The function to set the output PIN to low: setActiveState()

Add here as well:
void OpenTherm::end()
{
    detachInterrupt(digitalPinToInterrupt(inPin));
    setActiveState();  // Set output PIN low!
}

Also when you use OTA update, make sure that you end the OpenTherm and set output PIN low!

ArduinoOTA.onStart(displayOTAUpdate);

void displayOTAUpdate()
{
  ot.setActiveState();
  . . .
}

If I don't do that, my heater goes full pull when updating. Burning gas for nothing!

You can also apply this to WiFi-reconnect and ESP reset.