ihormelnyk/opentherm_library #71: Bit timing inaccurate with multi threading

ssharks, 2024-12-26:
The bit timing is arranged with delayMicroseconds(500);. On a single threaded application this works reasonable stable although rounding errors tend to accumulate.
As soon as multiple threads are used on an ESP32 or ESP8266, this is no longer guaranteed to take 500 us. Depending on the OS scheduling this causes some communication to work, some not to work.

Would it be better to use a timer for the bit timing? This guarantees the bit timing independent of scheduling and does not accumulate timing errors.

A workaround on the ESP32: Ensure all Opentherm communication is on 1 core, ensure the parrallel tasks are running on the other core.

Laxilef, 2024-12-30:
``delayMicroseconds()`` blocks the core, unlike ``delay()`` which calls ``vTaskDelay()``:

```c++
void ARDUINO_ISR_ATTR delayMicroseconds(uint32_t us) {
  uint64_t m = (uint64_t)esp_timer_get_time();
  if (us) {
    uint64_t e = (m + us);
    if (m > e) {  //overflow
      while ((uint64_t)esp_timer_get_time() > e) {
        NOP();
      }
    }
    while ((uint64_t)esp_timer_get_time() < e) {
      NOP();
    }
  }
}
```

ssharks, 2025-01-30:
Thank for the suggestion. vTaskDelay() does not work in this situation as the resolution for vTaskDelay() is either 1 or 10 ms, which prohibits from implementing a 0.5 ms delay.

A more accurate time is achievable using the timers of the ESP32, like with https://docs.espressif.com/projects/arduino-esp32/en/latest/api/timer.html

Laxilef, 2025-01-30:
``delayMicroseconds()`` does not call ``vTaskDelay()``.

ssharks, 2025-02-04:
I'm aware, delayMicroseconds expects full access to the processor. On the ESP32 multiple thread can be running on a single core. This means that a task can be interrupted and scheduled away. Therefore the delayMicroseconds can sometimes take 4 ms instead of 500 us. This breaks the functionality.

Laxilef, 2025-02-04:
For strict timing compliance and maintaining compatibility with all Arduino platforms, this is the optimal implementation, IMHO.
You can propose your own implementation for ESP32 by correcting the code with ``#ifdef ARDUINO_ARCH_ESP32`` and create a PR. From my experience Ihor is not against PR.

Phunkafizer, 2025-02-23:
I had this problem on ESP32 with higher network traffic (see image from logicanalyzer below), so I changed the TX routine using timer IRQs. I bit dirty though, suggestions welcome:

https://github.com/Phunkafizer/opentherm_library

[image]

ssharks, 2025-02-24:
> I had this problem on ESP32 with higher network traffic (see image from logicanalyzer below), so I changed the TX routine using timer IRQs. I bit dirty though, suggestions welcome:
> 
> https://github.com/Phunkafizer/opentherm_library
> 
> [image]

Look great, exactly what it should be. I've looked thought your code and I think it is just what I need. 

The construction with the txbuf I  would have made into a uint64_t / unsigned long in combination with some bitshifting. That would have made the code a little cleaner I think.

If you could make a pull request, it could maybe be merged.

ssharks, 2025-03-13:
> [@ssharks](https://github.com/ssharks) can you test [this](https://github.com/Laxilef/opentherm_library/tree/esp32_timer)?

I'll try to find a moment to test this soon.

ssharks, 2025-03-13:
> I had this problem on ESP32 with higher network traffic (see image from logicanalyzer below), so I changed the TX routine using timer IRQs. I bit dirty though, suggestions welcome:
> 
> https://github.com/Phunkafizer/opentherm_library
> 
> [image]

Are you OK if I transform it into a pull request?

Laxilef, 2025-03-13:
I tested Phunkafizer's implementation, it works too, but it introduces changes that break backward compatibility. Also, hwtimer is deprecated for idf 5, so I used gptimer in my implementation.

Before creating a PR, testing is necessary.

Phunkafizer, 2025-03-13:
> [@ssharks](https://github.com/ssharks) can you test [this](https://github.com/Laxilef/opentherm_library/tree/esp32_timer)?

I did a quick try but was not able to compile in platformio. The gptimer is not found, maybe because I'm using arduino framework.

ssharks, 2025-03-13:
> > [@ssharks](https://github.com/ssharks) can you test [this](https://github.com/Laxilef/opentherm_library/tree/esp32_timer)?
> 
> I did a quick try but was not able to compile in platformio. The gptimer is not found, maybe because I'm using arduino framework.

Hmmm... I pretty much get the opposite result. For me on the arduino framework the gptimer_handle_t is found, but the timerBegin(numTimer, 80, true) does have the wrong number of arguments.

I'm using arduino 1.8.19 and Arduino ESP32 2.0.18, @Laxilef can you share your version?

Laxilef, 2025-03-14:
I am using [pioarduino](https://github.com/pioarduino/platform-espressif32) latest stable version (Arduino core 3.1.3 based on IDF 5.3.2.250210).

ssharks, 2025-03-16:
I just tested with https://github.com/Laxilef/opentherm_library/tree/esp32_timer, commit 7160afe. It works like a charm.
While I was writing this message I discovered, you have added a few commit after that, which I did not include in the test.
I was able to run all tasks + loop on core 1, while still maintaining a reliable OpenTherm link. Many thanks for this fix.

Phunkafizer, 2025-03-16:
I change my platformio.ini to:

platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip
board = esp32-c3-devkitm-1
framework = arduino

before platform was "espressiv32"
But it seems that SOC_GPTIMER is not used:

[image]

What am I missing to be able to use SOC_GPTIMER?

Phunkafizer, 2025-03-16:
Update: Meanwhile I'm able to compile with laxilefs esp32 branch. I had to run a full clean before. But when I upload to my hardware I receive this via serial:

```
E (217) esp_image: Image length 1324672 doesn't fit in partition length 1310720
E (217) boot: OTA app partition slot 0 is not bootable
E (217) esp_image: image at 0x150000 has invalid magic byte (nothing flashed here?)
E (225) boot: OTA app partition slot 1 is not bootable
E (229) boot: No bootable app partitions in the partition table
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048b82
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1144
load:0x403cc710,len:0xad8
load:0x403ce710,len:0x2f80
entry 0x403cc710

```

My platformio settings are:

```
[env]
custom_version = 1.7
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
lib_compat_mode = strict
lib_ldf_mode = chain
board_build.filesystem = littlefs
board = esp32-c3-devkitm-1
framework = arduino
lib_deps = 
	ArduinoJson
	ESP32Async/AsyncTCP
  	ESP32Async/ESPAsyncWebServer
	https://github.com/marvinroger/async-mqtt-client
lib_ignore = 
	WebServer_WT32_ETH01
	WebServer_ESP32_ENC
	WebServer_ESP32_W5500
	WebServer_ESP32_W6100
build_flags = 
	-std=c++17
	-D BEARSSL_SSL_BASIC
	-D ARDUINO_USB_CDC_ON_BOOT=1
	-D ARDUINO_USB_MODE=1
	-D AP_SSID='"${secrets.ap_ssid}"'
	-D AP_PASSWORD='"${secrets.ap_password}"'
	-D HOSTNAME='"${secrets.hostname}"'
upload_speed = 921600
monitor_speed = 115200

```

Can anyone help me?

Laxilef, 2025-03-17:
> E (217) esp_image: Image length 1324672 doesn't fit in partition length 1310720

Use custom partition table, it looks like your firmware does not fit into partition. See example [[1]](https://github.com/Laxilef/OTGateway/blob/master/esp32_partitions.csv), [[2]](https://github.com/espressif/arduino-esp32/tree/master/tools/partitions), [docs](https://docs.platformio.org/en/latest/platforms/espressif32.html#partition-tables).

Phunkafizer, 2025-03-18:
Thank you so much, I wrote a custom partition table and it's working again now! 
Is it possible to remove "alwaysReceive" from constructor and making alwaysReceive weather public or accessible by a method? In my project I need to change this parameter according to mode (gateway / master / ...). I startet to implement OT smart power but would like to move on with your solution using gptimer which I like better than my testimplementation with hwtimer.

Laxilef, 2025-03-19:
I tried to stick to the style of the main repository. I think we should get @ihormelnyk approval to add a method for this because I would like to create a PR after testing.

ihormelnyk, 2025-03-19:
Sounds good. Please create PR compatible with Arduino and ESP platforms.

Phunkafizer, 2025-03-19:
Adding a new method or making alwaysReceive public should not make any breaking changes for backward compatibility.

Laxilef, 2025-03-20:
@Phunkafizer that's true, but we had to get approval from Ihor. Do a **Full clean** in platformio and try to compile your firmware, I added the ``OpenTherm::getAlwaysReceive()`` and ``OpenTherm::setAlwaysReceive(bool value)`` methods.

Phunkafizer, 2025-03-20:
It compiles and works from my side!

ssharks, 2025-03-30:
@Laxilef: After running with this library with commit https://github.com/ihormelnyk/opentherm_library/commit/7160afebba4a2ea655176965875a89e66282c85e I noted it started to loose connection more often after about one week of running. A reboot fixed this problem. This has now occurred 2 times. Do you have any idea what the root cause of this new issue might be?

Laxilef, 2025-03-30:
@ssharks Do you have any logs? What is the connection loss? What responses are you getting from the boiler?

ssharks, 2025-03-31:
At the moment I don't have any logs, as my device is stand alone and just gives an indication on its display if there is no connection to the boiler.
I can build in some logging if that would be convenient. I'll log the responses at least.

I-Connect, 2025-04-19:
Hi, I am running a setup that has several tasks running on an ESP32. It does MQTT over Ethernet, HTTP(S) put and get calls, reads a P1 port, handles BLE sensors/actuators and runs OpenTherm as Gateway (connected to a boiler and thermostat).

I am investigating issues where http(s) calls sometimes fail. These calls are time-consuming and will allow switching to other tasks in the process of connecting/sending/receiving. 

Troubleshooting/logging this is difficult as it is not directly reproducible (it happens maybe 10 times over 12 hours).

I do see an occasional invalid request being received from the Thermostat, and OpenTherm is a rather busy and externally triggered process where the thermostat fires off communications ~every second causing interrupts to be disabled/enabled  and messages are sent on a very slow baud rate so I was looking into this as it might contribute to the issues.

And I found this issue which could be somewhat related :-)

Not sure if it would resolve it but I am now testing a few hours with https://github.com/Laxilef/opentherm_library.git#esp32_timer. It runs fine and until now I have not gotten a http(s) or Open Therm failure yet. I will keep it running and let you know the results.